pulp-platform/ara #321: Similar issue as #320

Syedateeb24, 2024-07-07:
@mp-17 
i am facing an exactly similiar issue to #320 while executing the fmatmul.ideal program

cd fmatmul && if [ -d script ]; then python3 script/gen_data.py 128 128 128 > data.S ; else touch data.S; fi

sed -i s/"li t0, MSTATUS_FS | MSTATUS_XS$"/"li t0, MSTATUS_FS | MSTATUS_XS | MSTATUS_VS"/ /home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common/crt.S

git update-index --assume-unchanged /home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common/crt.S

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -DPREALLOCATE=1 -DSPIKE=1 -I/home/syed/Documents/ARA_new/ara/apps/riscv-tests/env -I/home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common -ffunction-sections -fdata-sections -c fmatmul/data.S -o fmatmul/data.S.o.spike

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -DPREALLOCATE=1 -DSPIKE=1 -I/home/syed/Documents/ARA_new/ara/apps/riscv-tests/env -I/home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common -ffunction-sections -fdata-sections -c /home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common/crt.S -o /home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common/crt.S.o.spike

mkdir -p bin/

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -Iinclude -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -DPREALLOCATE=1 -DSPIKE=1 -I/home/syed/Documents/ARA_new/ara/apps/riscv-tests/env -I/home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common -ffunction-sections -fdata-sections -o bin/fmatmul.spike fmatmul/kernel/fmatmul.c.o.spike fmatmul/main.c.o.spike fmatmul/data.S.o.spike /home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common/crt.S.o.spike /home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common/syscalls.c.o.spike common/util.c.o.spike -static -nostartfiles -lm -Wl,--gc-sections -nostdlib -T/home/syed/Documents/ARA_new/ara/apps/riscv-tests/benchmarks/common/test.ld -Wl,--gc-sections -DSPIKE

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/llvm-objdump --mattr=v -D bin/fmatmul.spike > bin/fmatmul.spike.dump

mkdir -p ideal_dispatcher/vtrace ideal_dispatcher/log ideal_dispatcher/temp

echo "run" | /home/syed/Documents/ARA_new/ara/install/riscv-isa-sim-mod/bin/spike --isa=rv64gcv_zfh --varch="vlen:4096,elen:64" -d bin/fmatmul.spike 2> ideal_dispatcher/temp/fmatmul.temp 1> ideal_dispatcher/log/fmatmul.log

cd ideal_dispatcher && scripts/vtrace.sh temp/fmatmul.temp vtrace/fmatmul.vtrace

The SPIKE log was not clean.

mkdir -p bin/

cp bin/fmatmul.spike bin/fmatmul.ideal``

and neither was i able to generate the fst waveform file , the output i got is:
`syed@syed:~/Documents/ARA_new/ara$ # Compile fmatmul for Ara                                                                      

make -C apps bin/fmatmul        

# Verilate the design

make -C hardware verilate trace=1 -B 

# Run fmatmul on Ara through Verilator

make -C hardware simv app=fmatmul trace=1 

make: Entering directory '/home/syed/Documents/ARA_new/ara/apps'

cd fmatmul && if [ -d script ]; then python3 script/gen_data.py 128 128 128 > data.S ; else touch data.S; fi

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c fmatmul/data.S -o fmatmul/data.S.o

chmod +x /home/syed/Documents/ARA_new/ara/apps/common/script/align_sections.sh

rm -f /home/syed/Documents/ARA_new/ara/apps/common/link.ld && cp /home/syed/Documents/ARA_new/ara/apps/common/arch.link.ld /home/syed/Documents/ARA_new/ara/apps/common/link.ld

/home/syed/Documents/ARA_new/ara/apps/common/script/align_sections.sh 4 /home/syed/Documents/ARA_new/ara/apps/common/link.ld

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c fmatmul/main.c -o fmatmul/main.c.o

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/crt0.S -o common/crt0-llvm.S.o

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/printf.c -o common/printf-llvm.c.o

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/string.c -o common/string-llvm.c.o

/home/syed/Documents/ARA_new/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/syed/Documents/ARA_new/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/serial.c -o common/serial-llvm.c.o

[... cut ...]
Makefile:83: "Specified QuestaSim version (questa-2021.2) not found in PATH /home/syed/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin"

build/verilator/Vara_tb_verilator -t -l ram,/home/syed/Documents/ARA_new/ara/apps/bin/fmatmul,elf

bash: line 1: build/verilator/Vara_tb_verilator: No such file or directory

make: *** [Makefile:231: simv] Error 127

make: Leaving directory '/home/syed/Documents/ARA_new/ara/hardware'`

Please help me resolve the issue.

mp-17, 2024-07-08:
Hello, 
I am sorry that there are issues with Verilator traces and ideal-dispatcher mode. I will need some time to debug it since we usually use QuestaSim to generate traces and in ideal-dispatcher mode

Syedateeb24, 2024-07-08:
@mp-17 
Based on the previous issues i have investigated i came across the solution for generating fst file is to bump the verilator version back to v4.214 which is mentioned in #260 
Does that really work ?
and tried to edit the make file in ```/ara ``` to ``` VERIL_VERSION           ?= v4.214 ``` and tried running ``` make verilator ``` and got the following error : 
[verilator_install_log.txt](https://github.com/user-attachments/files/16123786/verilator_install_log.txt)

Similar to the last query illustrated in #314