pulp-platform/ara #334: Any User Specific C program on ARA

Tanishqgithub, 2024-07-26:
I want to try and run any other different C programs on ARA for example the Binary sort in [https://www.geeksforgeeks.org/c-program-for-binary-insertion-sort/](program) 

What is the process to achieve this? 
Is it just to replace any ``` main.c ``` file in ```ara/apps/  ``` for example just replacing ``` main.c ``` file in ```apps/hello_world ``` ?

I sense some other way of writing a code to enable ARA processing or else it will just execute in ARIANE itself ? I am thinking correctly? 

If yes please guide me writing any user defined C code to be executed in ARA

grigohas, 2024-07-29:
Hello, have you found a solution? I want to do that too.

Tanishqgithub, 2024-07-29:
@grigohas No i haven't 
The authors have to give some idea regarding this

0rd1narY1, 2025-05-01:
Hello. I add my own programs in the following way. 
1. Create a folder in 'app/', such as 'sort'. 
2. Make sure that 'main.c' is in the root of 'sort/' and other '*.c' and '*.h' files in  'sort/kernel/'. Just like other apps. 
3. Then run 'make sort'. The program is compiled to folder 'bin/'.