pulp-platform/ara #335: error when I change config to 16 lanes

shkim-emily, 2024-07-29:
Hello, when I try to change Ara hardware configuration to 16_lanes error alert like below

my command is 
 cd hardware
make clean
make verilate config=16_lanes

Error is like below.
What is the problem and how can I solve the problem?

Makefile:83: "Specified QuestaSim version (questa-2021.2) not found in PATH /home/parallels/miniconda3/bin:/home/parallels/miniconda3/condabin:/home/parallels/intelFPGA/20.1/modelsim_ase/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin"
mkdir -p build
Successfully installed bender 0.27.3 in '/home/parallels/01_PRJ/00_ARA/ara/hardware'.
bender 0.27.3 available.
rm -rf build/verilator; mkdir -p build/verilator
/home/parallels/01_PRJ/00_ARA/ara/hardware/../hardware/bender script verilator -t rtl -t cv64a6_imafdcv_sv39 -t tech_cells_generic_include_tc_sram -t tech_cells_generic_include_tc_clk -t ara_test -t cva6_test -t verilator --define NR_LANES=16 --define VLEN=16384 --define ARIANE_ACCELERATOR_PORT=1 --define COMMON_CELLS_ASSERTS_OFF > build/verilator/bender_script_16_lanes
/home/parallels/01_PRJ/00_ARA/ara/install/verilator/bin/verilator -f build/verilator/bender_script_16_lanes           \
  -GNrLanes=16                                                         \
  -GVLEN=16384                                                                \
  -O3                                                                           \
  -Wno-fatal                                                                    \
  -Wno-PINCONNECTEMPTY                                                          \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LATCH                                                                    \
  -Wno-LITENDIAN                                                                \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  -Wno-ENUMVALUE                                                                \
  -Wno-COMBDLY                                                                  \
  tb/verilator/waiver.vlt                                                       \
  --Mdir build/verilator                                                       \
  -Itb/dpi                                                                      \
  --compiler clang                                                              \
  -CFLAGS "-DTOPLEVEL_NAME=ara_tb_verilator"                                        \
  -CFLAGS "-DNR_LANES=16"                                              \
  -CFLAGS -I/home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  ""                                                             \
  -LDFLAGS "-lelf"                                                              \
  ""                                                              \
  --exe                                                                         \
  /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
                                       \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
%Error: Verilator threw signal 9. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/parallels/01_PRJ/00_ARA/ara/install/verilator/bin/verilator_bin -f build/verilator/bender_script_16_lanes -GNrLanes=16 -GVLEN=16384 -O3 -Wno-fatal -Wno-PINCONNECTEMPTY -Wno-BLKANDNBLK -Wno-CASEINCOMPLETE -Wno-CMPCONST -Wno-LATCH -Wno-LITENDIAN -Wno-UNOPTFLAT -Wno-UNPACKED -Wno-UNSIGNED -Wno-WIDTH -Wno-WIDTHCONCAT -Wno-ENUMVALUE -Wno-COMBDLY tb/verilator/waiver.vlt --Mdir build/verilator -Itb/dpi --compiler clang -CFLAGS -DTOPLEVEL_NAME=ara_tb_verilator -CFLAGS -DNR_LANES=16 -CFLAGS -I/home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -CFLAGS -I/home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -CFLAGS -I/home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -LDFLAGS -lelf  --exe /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/dpi_memutil.cc /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/sv_scoped.cc /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/verilator_memutil.cc /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.cc /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc /home/parallels/01_PRJ/00_ARA/ara/hardware/tb/verilator/ara_tb.cpp --cc --top-module ara_tb_verilator
make: *** [Makefile:188: build/verilator/Vara_tb_verilator] Error 255

mp-17, 2024-08-22:
Hello @shkim-emily,

This is weird. Just to be 100% sure, have you applied the hardware patches as written in the readme?
Then, I would try to add `--debug --gdbbt` to the verilate command in the `hardware/Makefile` to see if we get some more information.

mp-17, 2024-08-22:
It can also be that [you are running out of memory](https://github.com/verilator/verilator/issues/2538) since the 16-lane design is huge.

mp-17, 2024-08-23:
This PR should help. Let me know if it works once it's merged!
https://github.com/pulp-platform/ara/pull/349

shkim-emily, 2024-09-05:
It works! 
Thank you so much :)

jracevedob, 2024-11-23:
@shkim-emily Hi Emily, did it work because you reduced the number of lanes to 2 or 4?