pulp-platform/ara #337: llama on ara

grigohas, 2024-07-30:
Hello, i want to simulate llama llm (the version on c++ that there is an option to cross compile it and simulate the riscv edition with vector extension)  on ara. I successfully installed all the tools and i run the examples from app folder. Do you thing that it is possible? Also, do you have any idea how i can do it?

aitesam961, 2024-08-08:
If you are able to cross-compile it for riscv assembly, you should be able to run it on ARA-SOC. The latest [RISCV-Toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) supports it but you need to build the toolchain for your selected ara config i.e specifying VLEN etc.
You need to be careful to ensure the resultant code is actually making use of vector instructions to gain all the advantage of vector

grigohas, 2024-08-08:
okay okay, and by "you should be able to run it on ara-soc" , you mean like the examples on app folder through verilator?

aitesam961, 2024-08-08:
There aren't any examples in app folder that by any way are relevant to
Llama AFAIK.

You will have to compile Llama for riscv assembly to be able to run on it.
Just like we can't run x86_64 applications on Apple M3 hardware.

grigohas, 2024-08-08:
yeah there is an option to cross compile llama to run on riscv with vector extension but i dont know which program to use to run it on ara-soc

aitesam961, 2024-08-08:
"Yeah, there is an option ...."
Can you point to it please?
Just want to know how you're planning crosscompile

grigohas, 2024-08-08:
when you do the make of llama you can add riscv_cross_compile=1 that does this 

```
ifdef RISCV_CROSS_COMPILE
CC	:= riscv64-unknown-linux-gnu-gcc
CXX	:= riscv64-unknown-linux-gnu-g++
endif
```

and riscv=1 that does this

```
else
	MK_CFLAGS   += -march=rv64gcv -mabi=lp64d
	MK_CXXFLAGS += -march=rv64gcv -mabi=lp64d
endif
```

aitesam961, 2024-08-08:
That's means it should be generating rv compatible binaries somewhere.
Can you check the resultant directories for it?

You can create a dump of the bins to check if it contains vector
instructions at all. To be safe at the first point.

Past that, you can modify the makefile to point it to your generated Llama
binary instead of the application generated by ara under /apps

It depends on how you plan to run that on ara. Either in simulations
(verilator/questasim) or on actual hardware (ara soc on fpga).

You may have to make changes to the makefile here and there to achieve the
output you want.

grigohas, 2024-08-08:
okay thank you very much . i am planning to run in simulations , i was trying verilator but without results

aitesam961, 2024-08-08:
github.com/aitesam961/ara-build-playground has a readme-I that may help you
resolve your issues with verilator.

grigohas, 2024-08-08:
do i have to change something to run the ara_soc or it is automated when i run an example of /apps folder?

aitesam961, 2024-08-09:
The only change you need is /ara/hardware/src/ara_soc.sv
Change Nrlanes and VLEN.
Nrlanes should be 2^n and Vlen should be 64*Nrlanes.

Other than that, you do not need any changes as far as you are running apps
from /apps folder.

grigohas, 2024-08-11:
okay but i have to specify a model for llama with -m option but the way that ara runs the examples from app folder doesnt have that option

aitesam961, 2024-08-11:
Like said earlier, I really do not think ara natively supports building
anything like Llama with its workflows.

All you need is a binary of your Llama application that is built with Ara
compatible riscv Toolchain.

So build it outside of ara, once you have it, place it under apps folder
and make changes in makefile to pick it up when called.

grigohas, 2024-08-14:
i changed the vlen and nrlanes at ara_soc but when i execute an app from app folder , does it call the ara_soc by default or i have to change that too?

aitesam961, 2024-08-14:
It does call the SoC which enables it to run the application.
You're fine with RTL as far as you configured the Nrlanes and Vlen
correctly.

Be sure to verify it's all working by running the simulation of any
available application for example

make -C apps bin/fmatmul
# Verilate the design
make -C hardware verilate trace=1 -B
# Run fmatmul on Ara through Verilator
make -C hardware simv app=fmatmul trace=1

grigohas, 2024-08-16:
okay but is it possible to run llama through verilator in general?

aitesam961, 2024-08-16:
Verilator simulates the RTL with the data loaded into testbenches and
responds with the data results that RTL generates. One way is waveform and
the other is text and data output generated by testbench assertions and
variables. The later seems feasible for you. You need to feed your prompt
to Llama binary that is running on ara and you will get the results from
verilator output in form of text that is generated by assertion variables.

If you are looking for a more realistic approach, you can run ara on for
example an FPGA, you need to run Llama binary on ara and you can tweak the
binary to take prompt from serial port and return result from the UART
which would be amazing.

grigohas, 2024-08-16:
i did the changes you suggested but when i run llama with the simv attribute i get a core dump but also the syntax to run llama is something like main -m (amodel).gguf but i cant pass the -m attribute with the simv option for ara with the following code

`.PHONY: simv
simv:
	$(veril_library)/V$(veril_top) $(if $(trace),-t,) -l ram,$(app_path)/$(app),elf`

aitesam961, 2024-08-16:
It will require more deeper digging. Perhaps someone from the community
might pick this issue up.

aitesam961, 2024-08-18:
Just found this
https://www.reddit.com/r/RISCV/comments/16qolk7/running_llamacpp_on_riscv_visionfive_2_simple/

So if you happen to get linux running on ara-soc somehow, the chances are
you'll be running Llama  on it.

For your info, the CVA6 core that is a part of ara-soc is capable of
booting Linux but in this case, you'll likely be building linux for this
soc yourself. CVA6 repo provided Linux image won't be compatible.

grigohas, 2024-08-18:
yeah i know , i found that yoo, thank you

closed 2024-09-18 by grigohas

KGback, 2025-03-17:
Hi, have you built bare-metal or linux environment of ara for running llama? I am building linux env for it, if possible, we can talk about it.