pulp-platform/ara #357: Some problem of Vivado

AD738560581, 2024-09-03:
Hello, I have encountered some issues regarding FPGA synthesis.  I conducted a comprehensive analysis of the entire ARA and found that it would terminate abnormally, and reported an abnormal termination(11) with Stack, which seems to be an issue of insufficient memory. But if the decoding module is annotated, it is found that it can start synthesis normally. In the end, I found that the decoding module's allways_comb was too large, causing Vivado to be unable to process it, resulting in a memory overflow. Anyone knows how to solve or avoid this problem？

AD738560581, 2024-09-04:
By the way, when we conducted a Lint check on the entire ARA, we found that spyglass reported that the always_comb process block cannot exceed 2000 lines, but the decoding module's always_comb process block has about 3000 lines. Have you encountered similar problems?

fulcrum34, 2024-09-06:
I have ARA-SOC synthesize successfully for Genesys2 (kitnex-7) FPGA. (4 lanes, 256 VLEN)
I handpicked the files from different sources, modified several sources and eventually got to a point where synth is successfull without critical warnings. Though there still seems to be a lot to fix.

[image]

Most of these warnings correspond to assertions which vivado doesn't support so I'm removing them. There are issues like case without default, unconnected nets, inferred latched and much more.
Will share the finished project under the @aitesam961 account.
Stay tuned

mrbilandi, 2024-09-09:
Hi @AD738560581 

I recommend using the latest version of Vivado. I also had some issues with synthesis, but after switching to Vivado 2024.1, it synthesized successfully.

fulcrum34, 2024-09-09:
Hi @mrbilandi , were you able to port it to an FPGA?
Any pointers to your way of collecting sources (I handpick files from deps). Any directions to xdc and running baremetal examples?

I am able to synthesize ara_soc but there are plenty of warnings (does not result in combinations logic, removing unused elements etc) which makes me think there are slim chances of it working.
Will surely try on Vivado 24 (The download is bigger than Empire state Building ( - : )