pulp-platform/ara #359: I am getting 0 as cycle count when doing the spike simulations

Ajsat3801, 2024-09-09:
Hi, 
I am facing a problem where the cycle count is always 0 when I am building an application and simulating with spike. I believe it's an issue on my side? Please help me solve this. PFA the screenshots I took for fft, dotproduct and fconv2d

PS: I am printing the values in hex for the fft as a workaround because it is displaying as %f otherwise, which was addressed in another issue.
Thanks and Regards

[image]
[image]
[image]

mp-17, 2024-10-17:
Hello @Ajsat3801. What do you need the runtime for? A runtime in SPIKE would not have much meaning; SPIKE is just a simulator that behaves like a processor with IPC == 1 and we use it to check the correctness of the software. 

Let me know!

Ajsat3801, 2024-10-17:
Hi @mp-17,
I realized that a few days back, and I figured out that I'll have to run the RTL simulations to get the runtime. I'm looking to get the cycle counts for FFT and a few other kernels, and honestly I'm a little lost. I am supposed to run something like this to get the cycle counts right? Is my understanding correct?

```                                                                     
make -C apps bin/fmatmul        
make -C hardware verilate trace=1 r
make -C hardware simv app=fmatmul trace=1 
```

Also I have been struggling with installing verilator. when i run make verilator, I'm getting this error

```
configure: error: in /ara/toolchain/verilator':
configure: error: C compiler cannot create executables
See `config.log' for more details
make: *** [Makefile:168: /ara/install/verilator] Error 77`
````

I found this workaround that allows the compiler to install it, by modifying the ara/Makefile

```
# Replace
CC=$(CLANG_CC) CXX=$(CLANG_CXX) CXXFLAGS=$(CLANG_CXXFLAGS) LDFLAGS=$(CLANG_LDFLAGS)

# With
CC="" 
```

but, I'm getting a new error which I am not able to resolve.

```
make[3]: Leaving directory '/ara/toolchain/verilator/src/obj_opt'
make[2]: Leaving directory '/ara/toolchain/verilator/src'
make[1]: Leaving directory '/ara/toolchain/verilator'
make: *** [Makefile:168: /ara/install/verilator] Error 2
```
Am I messing something up? Please help me resolve this