pulp-platform/ara #363: 'string.h' file not found when make bin/hello_world

hypertseng, 2024-10-09:
```
(base) root@88d591aaa016:/data1/zzx/ara/apps# make bin/hello_world
cd hello_world && if [ -d script ]; then python3 script/gen_data.py  > data.S ; else touch data.S; fi
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c hello_world/data.S -o hello_world/data.S.o
chmod +x /data1/zzx/ara/apps/common/script/align_sections.sh
rm -f /data1/zzx/ara/apps/common/link.ld && cp /data1/zzx/ara/apps/common/arch.link.ld /data1/zzx/ara/apps/common/link.ld
/data1/zzx/ara/apps/common/script/align_sections.sh 4 /data1/zzx/ara/apps/common/link.ld
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c hello_world/main.c -o hello_world/main.c.o
hello_world/main.c:20:10: fatal error: 'string.h' file not found
#include <string.h>
         ^~~~~~~~~~
1 error generated.
make: *** [/data1/zzx/ara/apps/common/runtime.mk:132: hello_world/main.c.o] Error 1
```

hypertseng, 2024-10-10:
After I re-build toolchain-llvm-newlib, /data1/zzx/ara/install/riscv-llvm/riscv64-unknown-elf appears, it indicates that the make process is not going as expected(toolchain-llvm-main=>toolchain-llvm-newlib=>toolchain-llvm-rt). So I add the include path manually with `export C_INCLUDE_PATH=/data1/zzx/ara/install/riscv-llvm/riscv64-unknown-elf/include`, then this problem is solved, but a new problem arises as follows:
`(base) root@88d591aaa016:/data1/zzx/ara/apps# make bin/hello_world
cd hello_world && if [ -d script ]; then python3 script/gen_data.py  > data.S ; else touch data.S; fi
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c hello_world/data.S -o hello_world/data.S.o
chmod +x /data1/zzx/ara/apps/common/script/align_sections.sh
rm -f /data1/zzx/ara/apps/common/link.ld && cp /data1/zzx/ara/apps/common/arch.link.ld /data1/zzx/ara/apps/common/link.ld
/data1/zzx/ara/apps/common/script/align_sections.sh 4 /data1/zzx/ara/apps/common/link.ld
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c hello_world/main.c -o hello_world/main.c.o
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/crt0.S -o common/crt0-llvm.S.o
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/printf.c -o common/printf-llvm.c.o
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/string.c -o common/string-llvm.c.o
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/serial.c -o common/serial-llvm.c.o
/data1/zzx/ara/install/riscv-llvm/bin/clang -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -c common/util.c -o common/util-llvm.c.o
mkdir -p bin/
/data1/zzx/ara/install/riscv-llvm/bin/clang -Iinclude -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/data1/zzx/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -o bin/hello_world hello_world/main.c.o hello_world/data.S.o common/crt0-llvm.S.o common/printf-llvm.c.o common/string-llvm.c.o common/serial-llv [... cut ...] c.o common/util-llvm.c.o -static -nostartfiles -lm -Wl,--gc-sections -T/data1/zzx/ara/apps/common/link.ld
ld.lld: error: unable to find library -lm
ld.lld: error: unable to find library -lc
ld.lld: error: unable to find library -lgloss
clang-16: error: ld command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:108: bin/hello_world] Error 1
rm common/util-llvm.c.o common/string-llvm.c.o common/crt0-llvm.S.o common/printf-llvm.c.o common/serial-llvm.c.o hello_world/main.c.o`

Azeem-Arshad, 2024-11-04:
@hypertseng Any update so far?

mp-17, 2024-11-28:
Hello @hypertseng,

I would first focus on installing the LLVM toolchain correctly. What's happening during the installation? You can try to share a screenshot of the errors.

raspbeep, 2025-02-14:
@Azeem-Arshad I managed to fix the linker error by adding this to the `apps/Makefile`:
`RISCV_LDFLAGS += -v -L{root}/install/riscv-llvm/riscv64-unknown-elf/lib -lm -lc -lgloss`
(replace the root by the path to the root dir of the cloned repo).
Although I do not know if this does not break something else in the future...

shawnnjupt, 2025-06-05:
I fix the problem in runtime.mk .
I add some code to find the library

```py
RISCV_CCFLAGS += -target riscv64-unknown-elf
RISCV_LDFLAGS += -L$(GCC_INSTALL_DIR)/lib/gcc/riscv64-unknown-elf/9.2.0
RISCV_LDFLAGS += -B${LLVM_INSTALL_DIR}/riscv64-unknown-elf/lib
RISCV_LDFLAGS += -L${LLVM_INSTALL_DIR}/riscv64-unknown-elf/lib
RISCV_LDFLAGS += -lc -lm -lgloss -lgcc

```