pulp-platform/ara #364: Error when make verilate

hanyjie, 2024-10-10:
```
$ make verilate
Makefile:83: "Specified QuestaSim version (questa-2021.2) not found in PATH /opt/riscv64gnu/bin:/opt/riscv64/bin:/home/hanyj/.vscode-server/cli/servers/Stable-38c31bc77e0dd6ae88a4e9cc93428cc27a56ba40/server/bin/remote-cli:/home/hanyj/.local/bin:/opt/riscv64gnu/bin:/opt/riscv64/bin:/home/hanyj/miniforge3/condabin:/home/hanyj/.cargo/bin:/home/hanyj/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
mkdir -p build
rm -rf build/verilator; mkdir -p build/verilator
/home/hanyj/Workspace/ara/hardware/../hardware/bender script verilator -t rtl -t cv64a6_imafdcv_sv39 -t tech_cells_generic_include_tc_sram -t tech_cells_generic_include_tc_clk -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define ARIANE_ACCELERATOR_PORT=1 --define COMMON_CELLS_ASSERTS_OFF > build/verilator/bender_script_default
/home/hanyj/Workspace/ara/install/verilator/bin/verilator -f build/verilator/bender_script_default           \
  -GNrLanes=4                                                         \
  -GVLEN=4096                                                                \
  -O3                                                                           \
  --hierarchical \
  -Wno-fatal                                                                    \
  -Wno-PINCONNECTEMPTY                                                          \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LATCH                                                                    \
  -Wno-LITENDIAN                                                                \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  -Wno-ENUMVALUE                                                                \
  -Wno-COMBDLY                                                                  \
  tb/verilator/waiver.vlt                                                       \
  --Mdir build/verilator                                                       \
  -Itb/dpi                                                                      \
  --compiler clang                                                              \
  -CFLAGS "-DTOPLEVEL_NAME=ara_tb_verilator"                                        \
  -CFLAGS "-DNR_LANES=4"                                              \
  -CFLAGS -I/home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  ""                                                             \
  -LDFLAGS "-lelf"                                                              \
  ""                                                              \
  --exe                                                                         \
  /home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/hanyj/Workspace/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
                                       \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/home/hanyj/Workspace/ara/hardware/build/verilator'
make[2]: Entering directory '/home/hanyj/Workspace/ara/hardware'
/home/hanyj/Workspace/ara/install/verilator/bin/verilator -f build/verilator/Vlane_c_hierMkArgs.f
sh: 1: exec: /home/hanyj/Workspace/ara/install/verilator/share/verilator/verilator_bin: not found
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/hanyj/Workspace/ara/install/verilator/share/verilator/verilator_bin -f build/verilator/Vlane_c_hierMkArgs.f
make[2]: *** [build/verilator/Vara_tb_verilator_hier.mk:336: hier_launch_verilator] Error 127
make[2]: Leaving directory '/home/hanyj/Workspace/ara/hardware'
make[1]: *** [Vara_tb_verilator_hier.mk:345: Vlane_c/lane_c.sv] Error 2
make[1]: Leaving directory '/home/hanyj/Workspace/ara/hardware/build/verilator'
%Error: make -C build/verilator -f Vara_tb_verilator_hier.mk   hier_verilation exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/hanyj/Workspace/ara/install/verilator/bin/verilator_bin -f build/verilator/bender_script_default -GNrLanes=4 -GVLEN=4096 -O3 --hierarchical -Wno-fatal -Wno-PINCONNECTEMPTY -Wno-BLKANDNBLK -Wno-CASEINCOMPLETE -Wno-CMPCONST -Wno-LATCH -Wno-LITENDIAN -Wno-UNOPTFLAT -Wno-UNPACKED -Wno-UNSIGNED -Wno-WIDTH -Wno-WIDTHCONCAT -Wno-ENUMVALUE -Wno-COMBDLY tb/verilator/waiver.vlt --Mdir build/verilator -Itb/dpi --compiler clang -CFLAGS -DTOPLEVEL_NAME=ara_tb_verilator -CFLAGS -DNR_LANES=4 -CFLAGS -I/home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -CFLAGS -I/home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -CFLAGS -I/home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -LDFLAGS -lelf  --exe /home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/dpi_memutil.cc /home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/sv_scoped.cc /home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/verilator_memutil.cc /home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.cc /home/hanyj/Workspace/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc /home/hanyj/Workspace/ara/hardware/tb/verilator/ara_tb.cpp --cc --top-module ara_tb_verilator
make: *** [Makefile:188: build/verilator/Vara_tb_verilator] Error 2
```
verilator version is `Verilator 5.012 2023-06-13 rev v5.012`

mp-17, 2024-10-17:
Hello @hanyjie, have you already checked [this](https://github.com/pulp-platform/ara/issues/314) out?

hanyjie, 2024-10-21:
I copied the verilator_bin from the bin directory to the /share/verilator directory and solved the problem. Thanks.