pulp-platform/ara #366: How to adding a custom RVV isa on ARA?

js4ngu, 2024-10-13:
Hey guys,
I'm a master student from Korea working on RVV custom ISA.

I am now trying to add my designed custom ISA to ara.
How can I add my custom ISA?

Is there anyone who can give me some good advice?
Because I have no idea how to go about it.
Sorry and thanks

Best regards

mp-17, 2024-10-17:
Hello @js4ngu, what do you need to do specifically? 

Usually, you can follow a basic prototype-oriented flow like:
1) Define a new instruction.
2) Find a free 32-bit encoding for that instruction following the RISC-V specifications.
3) Modify the decoders (CVA6's and Ara's) so that your instruction is recognized.
4) Add the necessary hardware to handle that instruction after the decoder.
5) Try to write the new instruction with [extended assembly](https://gcc.gnu.org/onlinedocs/gcc/Extended-Asm.html) and compile the .c program following our templates to quickly try it out.

js4ngu, 2024-10-18:
Thanks for the answer
I want to add a custom vector isa

Currently I have defined the custom instruction and have a riscv bin file containing it, which has been compiled to the compiler.

I hope to modify the ara to implement hw that supports it.

However, my custom hardware uses data from both neighbouring vector lanes.
Please see the image below I would like to start by designing a module that switches data from neighbouring vector lanes as shown below
[image]

I don't know how to modify it in this situation, can you give me a guide?

Also, I heard that in Ideal Dispatcher mode, cva6 treats it as a simple FIFO. Is it possible to work in my case without modifying the cva6 decoder, just modifying the ARA?