pulp-platform/ara #367: `make checkout` in hardware/ fails

jerryz123, 2024-10-24:
```
    Checkout cva6 (https://github.com/pulp-platform/cva6.git)
error: Git command (Command { std: cd "/nscratch/jerryz/chipyard-proj/chipyard-master2/generators/ara/ara/.bender/git/db/cva6-d9230a83a4be3a2a" && "git" "tag" "bender-tmp-5614fc9ceeae6db3851e8c721de868355db9476b" "5614fc9ceeae6db3851e8c721de868355db9476b" "--force", kill_on_drop: false }) in directory "/nscratch/jerryz/chipyard-proj/chipyard-master2/generators/ara/ara/.bender/git/db/cva6-d9230a83a4be3a2a" failed with exit code 128:

fatal: cannot update ref 'refs/tags/bender-tmp-5614fc9ceeae6db3851e8c721de868355db9476b': trying to write ref 'refs/tags/bender-tmp-5614fc9ceeae6db3851e8c721de868355db9476b' with nonexistent object 5614fc9ceeae6db3851e8c721de868355db9476b

make: *** [Makefile:138: checkout] Error 1
```

0rd1narY1, 2024-11-01:
Hello! I met the same problem. Do you know how to solve it?

mp-17, 2024-11-25:
Hey @jerryz123, sorry for that! Is it still the case? I should have fixed this.

JonasReichhardt, 2025-02-24:
Hey @mp-17, the same error occurred when using the current main branch (commit hash bfb6439)

mp-17, 2025-02-26:
Hey @JonasReichhardt, weird, I can checkout to CVA6's `feb5f72b5cafb840388ea85b155dfb4013c1926f` [commit](https://github.com/pulp-platform/cva6/commits/mp/pulp-v1-araOS/). Have you tried to clean up both `.bender` and `hardware/deps` folders?