pulp-platform/ara #389: cva6-sdk host key verification failed on git submodule update --init --recursive

rseac, 2025-01-30:
Hello,

I get the following error when running `git submodule update --init --recursive`.

```
Cloning into '/ara/cheshire/sw/cva6-sdk'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:moimfeld/cva6-sdk.git' into submodule path '/ara/cheshire/sw/cva6-sdk' failed
```

I believe the problem is with the `.gitsubmodules` url for `cheshire/sw/cva6-sdk`.  It currently has 
```
url = git@github.com:moimfeld/cva6-sdk.git
```
but from the history it seems like it should be
```
url = https://github.com/pulp-platform/cva6.git
```

Is this correct?  I'd be happy to open a PR, if this is the case.

mp-17, 2025-02-04:
Hello @rseac,

Sorry for leaving your question behind.
It should be `cva6-sdk` if we are talking about `ara/cheshire/sw/cva6-sdk`. It's weird since the pointer seems correct to `https://github.com/moimfeld/cva6-sdk/tree/mi/cheshire-ara-os`. Anyway, we will bump this soon. In the meantime, you can try to clone the SDK directly. The commit is correct (`ed582f3b8ea78a3d0a20f81b3da0f36a73d7e86b`). I will also try to take a closer look tonight or tomorrow. Let me know if cloning directly works for you as a temporary fix :-)

rseac, 2025-03-06:
Hi @mp-17,

Cloning directly seems to be fine.  I am using that in my docker images for now.

Thanks.