pulp-platform/ara #393: Boot linux on FPGA

KGback, 2025-02-14:
Hello, thank you for your work on ara! I want to use ara and cheshire to boot linux on FPGA vcu128 for running llama.cpp, what commits of cheshire and ara is stable?

Can I use branch main of ara and branch mp/ara-pulpv2-os-rebase of cheshire?

KGback, 2025-03-17:
Hey, I have checkout to `mp/ara-pulpv2-os-rebase`, but when i run `bender checkout` after checkout, there is one error.

```
error: Failed to checkout commit e977bb1be465c27a2a2b8cbc34830b4796af9ccf for ara given in Bender.lock.
 Git command (Command { std: cd "/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/.bender/git/db/ara-55c77ee6cf862ad0" && "git" "tag" "bender-tmp-e977bb1be465c27a2a2b8cbc34830b4796af9ccf" "e977bb1be465c27a2a2b8cbc34830b4796af9ccf" "--force", kill_on_drop: false }) in directory "/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/.bender/git/db/ara-55c77ee6cf862ad0" failed with exit code 128:

fatal: cannot update ref 'refs/tags/bender-tmp-e977bb1be465c27a2a2b8cbc34830b4796af9ccf': trying to write ref 'refs/tags/bender-tmp-e977bb1be465c27a2a2b8cbc34830b4796af9ccf' with nonexistent object e977bb1be465c27a2a2b8cbc34830b4796af9ccf

```

Could you please guide me on how to fix it?

KGback, 2025-03-18:
Hey, @mp-17 , for that problem of repo missing, I used branch `mp/pulp-v2-os-fpga` of ara, which commit is 1726fec. But when I run `make ara-chs-xilinx`, errors occurs like this: 

```
ERROR: [Synth 8-9263] cannot open include file 'ara/intf_typedef.svh' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:115]
ERROR: [Synth 8-10157] use of undefined macro 'CVA6_TYPEDEF_EXCEPTION' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:592]
ERROR: [Synth 8-2716] syntax error near '(' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:592]
ERROR: [Synth 8-8891] 'Cva6Cfg' is already declared [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:592]
INFO: [Synth 8-6826] previous declaration of 'Cva6Cfg' is from here [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:561]
ERROR: [Synth 8-10157] use of undefined macro 'CVA6_INTF_TYPEDEF_ACC_REQ' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:594]
ERROR: [Synth 8-8891] 'Cva6Cfg' is already declared [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:594]
INFO: [Synth 8-6826] previous declaration of 'Cva6Cfg' is from here [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:561]
ERROR: [Synth 8-2716] syntax error near 'roundmode_e' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:594]
ERROR: [Synth 8-2716] syntax error near ')' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:594]
ERROR: [Synth 8-10157] use of undefined macro 'CVA6_INTF_TYPEDEF_ACC_RESP' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:595]
ERROR: [Synth 8-8891] 'Cva6Cfg' is already declared [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:595]
INFO: [Synth 8-6826] previous declaration of 'Cva6Cfg' is from here [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:561]
ERROR: [Synth 8-2716] syntax error near ')' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:595]
ERROR: [Synth 8-10157] use of undefined macro 'CVA6_INTF_TYPEDEF_MMU_REQ' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:597]
ERROR: [Synth 8-8891] 'Cva6Cfg' is already declared [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:597]
INFO: [Synth 8-6826] previous declaration of 'Cva6Cfg' is from here [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:561]
ERROR: [Synth 8-10157] use of undefined macro 'CVA6_INTF_TYPEDEF_MMU_RESP' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:598]
ERROR: [Synth 8-8891] 'Cva6Cfg' is already declared [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:598]
INFO: [Synth 8-6826] previous declaration of 'Cva6Cfg' is from here [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:561]
ERROR: [Synth 8-2716] syntax error near ')' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:598]
ERROR: [Synth 8-10157] use of undefined macro 'CVA6_INTF_TYPEDEF_CVA6_TO_ACC' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:600]
ERROR: [Synth 8-2716] syntax error near ')' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:600]
ERROR: [Synth 8-10157] use of undefined macro 'CVA6_INTF_TYPEDEF_ACC_TO_CVA6' [/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv:601]
ERROR: [Synth 8-12188] Failed to read verilog '/home/data/projects/cpu_projs/cheshire_ara-pulpv2-os-rebase/hw/cheshire_soc.sv'
```
Are some key files missing? 

By the way, I also tried branch `mp/ara-pulpv1-os` of cheshire and succeed to run zsbl and firmware on FPGA, but it stopped at `opensbi`. I used ila to capture pc of cva6, found it fell into a loop of `spin_lock` in `sbi_hart_init`.

```
/\___/\       Boot mode:       2                                            
( o   o )      Real-time clock: 1000000 Hz                                   
(  =^=  )      System clock:    50080000 Hz                                  
(        )     Read global ptr: 0x02001b2c                                   
(    P    )    Read pointer:    0x02000bc7                                   
(  U # L   )   Read argument:   0x1001ffa0                                   
(    P      )                                                                
(           ))))))))))                                                       
                                                                             
[ZSL] Copy device tree (part 1, LBA 128-159) to 0x80800000... OK             
[ZSL] Copy firmware (part 2, LBA 2048-8191) to 0x80000000... OK              
[ZSL] Launch firmware at 80000000 with device tree at 80800000
```
I have no idea about it, Do you have any suggestions?

mp-17, 2025-03-18:
Hey @KGback,

While waiting for the release, you can try [this branch](https://github.com/pulp-platform/cheshire/tree/mp/ara-pulp-v2) following the example script you can find [here](https://github.com/pulp-platform/ara/tree/main/cheshire). 
It's still work in progress so some commits can break; if you want a commit that does not break, you can modify Cheshire's Bender.lock commit for Ara to `05c1616d2317d2fda09b3fdf69e8aae4c2e55eaf`. This is on `main` and should not break.

KGback, 2025-03-19:
OK, Thank you very much, I will try this branch.