pulp-platform/ara #397: Do examples with intrinsics need to be updated (i.e. dot product)?

rseac, 2025-02-25:
I am using clang 19 with RVV intrinsics 1.0.  I noticed that the examples that use intrinsics do not use the `__riscv_` prefix.  For example, apps/dotproduct does not use the prefix.  Do these need to be updated?

mp-17, 2025-02-26:
Hey @rseac,

Yes, I actually need to update all the intrinsics. I plan to bump the compilers and the intrinsics together, and I will likely converge on GCC only to simplify the toolchain management in the repo. It's in the pipeline :-)

mp-17, 2025-03-04:
Bumping everything [here](https://github.com/pulp-platform/ara/pull/402)

mp-17, 2025-03-19:
The current PR with an LLVM bump is [here](https://github.com/pulp-platform/ara/pull/404) now since the newer gcc compilers seem not able to compile RVV inline assembly without adding any new vector instruction (as required by our legacy riscv-tests).

mp-17, 2025-03-20:
[PR](https://github.com/pulp-platform/ara/pull/404) merged. Feel free to re-open if there are issues!

closed 2025-03-20 by mp-17