pulp-platform/ara #400: How is the Max Performance estimated?

LorcaQAQ, 2025-03-04:
I am reading your manuscript on Ara2, but I have a question about your Table 2, How do you estimate the Max Performance? Could you provide some examples to illustrate the calculation? Thanks!

mp-17, 2025-03-04:
Hello @LorcaQAQ,

I think you refer to the max. performance calculated to see the performance ideality on Ara2, i.e., how ideal is the performance on the maximum achievable by that kernel on the vector processor. So, max. performance is the maximum number of operations per cycle achievable, limited by the number of FPUs if the kernel is memory bound on Ara2, or limited by the memory BW if the kernel is memory bound on Ara2. 
Also, since we talk about operations, FMACC counts as 2 operations while FADD counts as one operation. So, if there are only FMACC and the kernel is computation bound, we have max 2 op/cycle in terms of throughput. If they are only FADD, we have 1 op/cycle max. If there is an intermediate situation, we need to calculate a weighted average. Let's suppose to only have 1 lane for the following examples:

```
loop:
 vfadd
 vfadd
```
Max 1 OP/cycle

```
loop:
 vfmacc
 vfadd
```
Max 1.5 OP/cycle

```
loop:
 vfmacc
 vfmacc
```
Max 2 OP/cycle

When the kernel is memory bound, the max. performance for performance ideality is limited by the memory bandwidth (in Ara, the memory BW is 32L-bit/cycle, while the compute works at 64L-bit/cycle).
```
loop:
 vld
 vld
 vadd
```
Max 0.25 OP/cycle

Hope it helps!

LorcaQAQ, 2025-03-04:
Thanks! I think I now understand the four examples. But I still have a question. In Table 2, under the first row for "matmul," what does "1 ×" represent in the expression "1 × 2.0 × L"? Additionally, in the "dropout" row, what does "2 ×" stand for?