pulp-platform/ara #408: Errors with building toolchain-llvm

zhoukai2222, 2025-04-04:
Hello
When I used the command `make toolchain-llvm` to compile the toolchain, the process was ultimately interrupted and output the following errors. 
```
[4/344] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o
FAILED: lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o 
/home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/clang++ -DHAVE_RPC_XDR_H=0 -D__func__=__FUNCTION__ -I/home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/.. -O3 -std=c++17 -MD -MT lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o -MF lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o.d -o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o -c /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_deadlock_detector1.cpp
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_deadlock_detector1.cpp:13:
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_deadlock_detector_interface.h:22:
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_internal_defs.h:15:
/home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_platform.h:18:4: error: "This operating system is not supported"
   18 | #  error "This operating system is not supported"
      |    ^
1 error generated.
ninja: build stopped: subcommand failed.
Makefile:98: recipe for target 'toolchain-llvm-rt' failed
make: *** [toolchain-llvm-rt] Error 1
```
The terminal output is shown below. Could you please suggest any solutions?
The complete output log file is also provided here:

```
cd /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt && rm -rf build && mkdir -p build && cd build && \
cmake /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt -G Ninja \
-DCMAKE_INSTALL_PREFIX=/home/zk/Ariane_Ara/ara/install/riscv-llvm \
-DCMAKE_C_COMPILER_TARGET="riscv64-unknown-elf" \
-DCMAKE_ASM_COMPILER_TARGET="riscv64-unknown-elf" \
-DCOMPILER_RT_DEFAULT_TARGET_ONLY=ON \
-DCOMPILER_RT_BAREMETAL_BUILD=ON \
-DCOMPILER_RT_BUILD_BUILTINS=ON \
-DCOMPILER_RT_BUILD_LIBFUZZER=OFF \
-DCOMPILER_RT_BUILD_MEMPROF=OFF \
-DCOMPILER_RT_BUILD_PROFILE=OFF \
-DCOMPILER_RT_BUILD_SANITIZERS=OFF \
-DCOMPILER_RT_BUILD_XRAY=OFF \
-DCMAKE_C_COMPILER_WORKS=1 \
-DCMAKE_CXX_COMPILER_WORKS=1 \
-DCMAKE_SIZEOF_VOID_P=4 \
-DCMAKE_C_COMPILER="/home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/clang" \
-DCMAKE_C_FLAGS="-march=rv64gc -mabi=lp64d -mno-relax -mcmodel=medany" \
-DCMAKE_ASM_FLAGS="-march=rv64gc -mabi=lp64d -mno-relax -mcmodel=medany" \
-DCMAKE_AR=/home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/llvm-ar \
-DCMAKE_NM=/home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/llvm-nm \
-DCMAKE_RANLIB=/home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/llvm-ranlib \
# -DLLVM_CONFIG_PATH=/home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/llvm-config
-- The C compiler identification is Clang 20.1.0
-- The CXX compiler identification is Clang 20.1.0
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/clang
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for unwind.h
-- Looking for unwind.h - not found
-- Looking for rpc/xdr.h
-- Looking for rpc/xdr.h - not found
-- Performing Test COMPILER_RT_HAS_NO_DEFAULT_CONFIG_FLAG
-- Performing Test COMPILER_RT_HAS_NO_DEFAULT_CONFIG_FLAG - Failed
CMake Warning at cmake/Modules/CompilerRTUtils.cmake:335 (message):
  LLVMTestingSupport not found in LLVM_AVAILABLE_LIBS
Call Stack (most recent call first):
  CMakeLists.txt:110 (load_llvm_config)

-- LLVM_MAIN_SRC_DIR: "/home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/llvm"
-- Found Python3: /usr/bin/python3.8 (found version "3.8.0") found components: Interpreter 
-- cmake c compiler target: riscv64-unknown-elf
[... 246 lines cut ...]
FAILED: lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_allocator.cpp.o 
/home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/clang++ -DHAVE_RPC_XDR_H=0 -D__func__=__FUNCTION__ -I/home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/.. -O3 -std=c++17 -MD -MT lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_allocator.cpp.o -MF lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_allocator.cpp.o.d -o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_allocator.cpp.o -c /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_allocator.cpp
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_allocator.cpp:14:
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_allocator.h:16:
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_common.h:18:
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_flags.h:16:
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_internal_defs.h:15:
/home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_platform.h:18:4: error: "This operating system is not supported"
   18 | #  error "This operating system is not supported"
      |    ^
1 error generated.
[4/344] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o
FAILED: lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o 
/home/zk/Ariane_Ara/ara/install/riscv-llvm/bin/clang++ -DHAVE_RPC_XDR_H=0 -D__func__=__FUNCTION__ -I/home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/.. -O3 -std=c++17 -MD -MT lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o -MF lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o.d -o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.riscv64.dir/sanitizer_deadlock_detector1.cpp.o -c /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_deadlock_detector1.cpp
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_deadlock_detector1.cpp:13:
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_deadlock_detector_interface.h:22:
In file included from /home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_internal_defs.h:15:
/home/zk/Ariane_Ara/ara/toolchain/riscv-llvm/compiler-rt/lib/sanitizer_common/sanitizer_platform.h:18:4: error: "This operating system is not supported"
   18 | #  error "This operating system is not supported"
      |    ^
1 error generated.
ninja: build stopped: subcommand failed.
Makefile:98: recipe for target 'toolchain-llvm-rt' failed
make: *** [toolchain-llvm-rt] Error 1
```

zhoukai2222, 2025-04-04:
[output.log](https://github.com/user-attachments/files/19604387/output.log)

zhoukai2222, 2025-04-04:
The complete output log file is provided here:

[output.log](https://github.com/user-attachments/files/19604408/output.log)

ferchup, 2025-04-08:
Hello @zhoukai2222 and @TeoThatsMe , same problem here. If you find the solution could you please post it?