pulp-platform/ara #411: Problem of FPU support

AD738560581, 2025-04-15:
Hello@mp-17 @suehtamacv , there is a problem that we want to support ZVE64f for RISCV RVV, which means there is no support for FP16 and FP64. Therefore, we configured the FPUSupport for ara as FPUSupportSingle. However, when we compile certain complex applications using -rv64gc_zve64f, vfncvt and vfwcvt instructions appear, which on the spec indicate that converting SEW16 integers to FP32 data seems make sense. However, it seems that ARA is currently unable to support this scenario. I would like to ask how to effectively avoid the generation of this command？

mp-17, 2025-06-12:
Hi @AD738560581, this is [a bug](https://github.com/pulp-platform/ara/blob/main/hardware/src/ara_dispatcher.sv#L2566) in the illegal insn check of Ara, thanks a lot for reporting. As a quick workaround, if you are sure your code is legal, you can just delete that check (actually, you can remove all the legality checks in the dispatcher related to `FPUSupportSingle`).

mp-17, 2025-06-12:
You are right, will fix this as well. 
As a quick fix, you can modify [this line](https://github.com/pulp-platform/ara/blob/main/hardware/src/lane/vmfpu.sv#L851) to 

```
IntFmtMask : {1'b1, 1'b1, 1'b1, 1'b1}`.
```

 Let me know if there are still issues after this modification.