pulp-platform/ara #416: make toolchain-llvm failing

malifarooq, 2025-05-31:
Hi, `make toolchain-llvm` is failing for me. I tried the solution suggested in #410 but it did not fix the issue. I get the error "This operating system is not supported"

`[421/592] Building ASM object lib/hwasan/CMakeFiles/RTHwasan.riscv64.dir/hwasan_setjmp_riscv64.S.o
FAILED: lib/hwasan/CMakeFiles/RTHwasan.riscv64.dir/hwasan_setjmp_riscv64.S.o 
/mnt/vault1/mfaroo19/internship/ara/install/riscv-llvm/bin/clang -target riscv64-unknown-elf -DHWASAN_WITH_INTERCEPTORS=1 -D__func__=__FUNCTION__ -I/mnt/vault1/mfaroo19/internship/ara/toolchain/riscv-llvm/compiler-rt/lib/hwasan/.. -march=rv64gc -mabi=lp64d -mno-relax -mcmodel=medany -fPIC -fno-builtin -fno-exceptions -fomit-frame-pointer -funwind-tables -fno-stack-protector -fvisibility=hidden -fno-lto -O3 -g -Wno-variadic-macros -nostdinc++ -fno-rtti -Wno-format -MD -MT lib/hwasan/CMakeFiles/RTHwasan.riscv64.dir/hwasan_setjmp_riscv64.S.o -MF lib/hwasan/CMakeFiles/RTHwasan.riscv64.dir/hwasan_setjmp_riscv64.S.o.d -o lib/hwasan/CMakeFiles/RTHwasan.riscv64.dir/hwasan_setjmp_riscv64.S.o -c /mnt/vault1/mfaroo19/internship/ara/toolchain/riscv-llvm/compiler-rt/lib/hwasan/hwasan_setjmp_riscv64.S
In file included from /mnt/vault1/mfaroo19/internship/ara/toolchain/riscv-llvm/compiler-rt/lib/hwasan/hwasan_setjmp_riscv64.S:18:
/mnt/vault1/mfaroo19/internship/ara/toolchain/riscv-llvm/compiler-rt/lib/hwasan/../sanitizer_common/sanitizer_platform.h:18:4: error: "This operating system is not supported"
`

malifarooq, 2025-05-31:
Reverting to commit #05c1616 lets `make toolchain-llvm` work. Maybe the issue is in the update to LLVM v20.1.0?

vaibhavmindgrove22, 2025-10-23:
I ran into the same problem. Has anyone here managed to fix it?

3fireworks, 2025-12-02:
The same problem. The OS was Macos Tahoe and Rocky9 Linux.