pulp-platform/ara #419: vwmacc.vx error when rs1 is Temporary registers

hypertseng, 2025-06-11:
my application is a gemm kernel with int8 input and int16 output, partial dumped codes is as follows:
`8007be8a: f727e057     	vwmacc.vx	v0, a5, v18
8007be8e: 00058783     	lb	a5, 0x0(a1)
8007be92: f72f6157     	vwmacc.vx	v2, t5, v18
8007be96: 95b6         	add	a1, a1, a3
8007be98: 00058f03     	lb	t5, 0x0(a1)
8007be9c: f72ee257     	vwmacc.vx	v4, t4, v18`

the instruction which PC is 8007be8a can execute normally, but the 8007be92 instruction throws an exception:
`Exception @  65601500, PC: 000000008007be92, Cause: Illegal Instruction, 
				tval: 00000000f72f6157`

mp-17, 2025-06-12:
Hi @hypertseng, can you please share the current `vtype` and `vl` configuration? Which is the last `vsetvl`, `vsetivli`, `vsetvli`, or CSR instruction executed before this code fragment?

mp-17, 2025-06-12:
Are you sure your code is correct, e.g., by running it on SPIKE or a golden simulator? 
The issue is probably the vector register, and not the scalar register.

```
Some vector instructions have source and destination vector operands with the same number of
elements but different widths, so that EEW and EMUL differ from SEW and LMUL respectively but
EEW/EMUL = SEW/LMUL. For example, most widening arithmetic instructions have a source group
with EEW=SEW and EMUL=LMUL but have a destination group with EEW=2*SEW and EMUL=2*LMUL.

...

# Vector Widening Integer Multiply-Add Instructions
The widening integer multiply-add instructions add the full 2*SEW-bit product from a SEW-bit*SEW-bit multiply to a 2*SEW-bit value and produce a 2*SEW-bit result. All combinations of signed and
unsigned multiply operands are supported.
```

In your case, it seems that EMUL of the destination register is `EMUL = EEW * LMUL/SEW = 16 * 2/8 = 4`. If this is the case, `v2` as destination register is illegal.

hypertseng, 2025-06-12:
Oh thank you very much, it's indeed the application's error.

closed 2025-06-12 by hypertseng